bigtreetech/Octopus-Max-EZ #1: Unable o upload firmware

gsuresh2u, 2023-02-21:
Hi, i installed klipper in Pi, then generated klipper.bin file and downloaded, renamed it to firmware.bin
after inserttting the file to sd card, it uploaded fine (file name changed to FIRMWARE.CUR)
But klipper not connected witth msu, i am getign mcu serial fine
then tried to flash firmware again with other klipper.bin
then its not uplaoding firmware, file name remains the same
then i tried to connected to pc, then it showing as COM6
then i tried to upload Marlin firmware directly from VS Code, then its saying "unable to find a matching cmsis-dap device"
is there any problem witth bootloader ?
there are 3 buttons on board, RESET, BOOT0, BOOT1, is there any way to reset o facory defaults with those buttons ?
and how to enter DFU mode ? to upload firmware directly from PI using DFU mode ?
i tried with "make flash FLASH_DEVICE=1d50:614e"
that 1d50:614e is showinf as usb serial in putty with command "lsusb"
[image]
[image]

gsuresh2u, 2023-02-22:
@bigtreetech, please help me to reset board to factory default

gsuresh2u, 2023-02-23:
It's ok

gsuresh2u, 2023-02-23:
Close

closed 2023-02-26 by gsuresh2u

KumphanartDansiri, 2023-06-07:
@gsuresh2u Do you able to reset the board to factory default?  After flashing firmware.bin, I am no longer see the serial and want to reset it back.

KumphanartDansiri, 2023-06-07:
@gsuresh2u Can you provide your klipper firmware?