bigtreetech/Octopus-Max-EZ #2: Bl touch pin

Ipermax2000, 2023-03-08:
Attention in version V 1.0 of Octopus Max Ex, you have to reverse the connection of the two pins PB14 and PB!% inside printer.cfg. Otherwise the BL Touch does not work. I write the exact code here:

```

[bltouch]
sensor_pin: ^PB15
#   Pin connected to the BLTouch sensor pin. Most BLTouch devices
#   require a pullup on the sensor pin (prefix the pin name with "^").
#   This parameter must be provided.
control_pin: PB14
#   Pin connected to the BLTouch control pin. This parameter must be
#   provided.
pin_move_time: 0.680
#   The amount of time (in seconds) to wait for the BLTouch pin to
#   move up or down. The default is 0.680 seconds.
stow_on_each_sample: True
#   This determines if Klipper should command the pin to move up
#   between each probe attempt when performing a multiple probe
#   sequence. Read the directions in docs/BLTouch.md before setting
#   this to False. The default is True.
probe_with_touch_mode: False
#   If this is set to True then Klipper will probe with the device in
#   "touch_mode". The default is False (probing in "pin_down" mode).
pin_up_reports_not_triggered: True
#   Set if the BLTouch consistently reports the probe in a "not
#   triggered" state after a successful "pin_up" command. This should
#   be True for all genuine BLTouch devices. Read the directions in
#   docs/BLTouch.md before setting this to False. The default is True.
#pin_up_touch_mode_reports_triggered: True
#   Set if the BLTouch consistently reports a "triggered" state after
#   the commands "pin_up" followed by "touch_mode". This should be
#   True for all genuine BLTouch devices. Read the directions in
#   docs/BLTouch.md before setting this to False. The default is True.
set_output_mode: 5V
#   Request a specific sensor pin output mode on the BLTouch V3.0 (and
#   later). This setting should not be used on other types of probes.
#   Set to "5V" to request a sensor pin output of 5 Volts (only use if
#   the controller board needs 5V mode and is 5V tolerant on its input
#   signal line). Set to "OD" to request the sensor pin output use
#   open drain mode. The default is to not request an output mode.
x_offset: 0
y_offset: -24.047
Z_offset: 2.550
speed: 2.0
lift_speed: 2.0
samples: 2
sample_retract_dist: 3
samples_result: average
samples_tolerance:0.055
samples_tolerance_retries: 0
#   See the "probe" section for information on these parameters.

Following I write the wrong code that you find from the manufacturer:

#[bltouch]
#sensor_pin: PB14
#control_pin: PB15

gsuresh2u, 2023-03-09:
Yes, I already found out 2 days back
Thanks for confirmation

Ipermax2000, 2023-03-09:
okay

closed 2023-04-01 by Ipermax2000