bigtreetech/Octopus-Max-EZ #4: schematic diagram

22chrs, 2023-04-20:
Please release the schematic like you have for your other boards. It's really helpful for troubleshooting issues. I want to use several boards for an industrial application, where we want to develop custom firmware – for that purpose it would also be very helpful to have the schematics. 
Thanks a lot!

Freddyb888, 2023-04-20:
Following

idontknowtom, 2023-05-20:
@bigtreetech Many people on the Facebook page are asking for the Octopus Max EZ schematic. Can this be posted on GitHub soon? 

Thanks

lianghongkey, 2023-05-26:
Following

Freddyb888, 2023-05-26:
Following. I want it too.!!

22chrs, 2023-05-29:
@bigtreetech Could you say if this is coming or any update? That would be very helpful. Thanks a lot!

Freddyb888, 2023-06-13:
I paid over 100 euro for this board and they cannot give the complete documentation?? I will make work of this, also complain to amazon, 3djake,.....they don't even bother to answer a few questions. What are you busy counting money?.??????

ellensp, 2023-06-26:
These have been added https://github.com/bigtreetech/Octopus-Max-EZ/blob/master/Hardware/BIGTREETECH%20Octopus%20MAX%20EZ%20V1.0-SCH.pdf

jonathanperret, 2023-06-26:
I wanted to look at the schematic to understand the USB-C/USB-A situation — in particular why powering the board through USB-C didn't work with "smart" cables.

The schematic does show that this board has the same issue as the original revision of the Raspberry Pi 4, i.e. tying the two CC lines together (https://www.scorpia.co.uk/2019/06/28/pi4-not-working-with-some-chargers-or-why-you-need-two-cc-resistors/). Oh well.

On the other hand, I'm surprised by the value of the CC resistor (R191) mentioned in the schematic, which is 1M when everywhere says to use 5.1k for CC resistors on a device.

[image]

I don't have my Octopus board handy to check right now, but if this is actually a 1M resistor and not a schematic error, it looks like we're lucky hosts will power the board at all, I guess?