bigtreetech/Octopus-Max-EZ #5: Octopus Max not working with EZ31865

Michael-Jacobsen, 2023-04-23:
I'm using an Octopus Max with the EZ31865 designed for it. With the EZ31865 on Motor8 slot connected to a PT1000 and everything configured correctly in klipper it works just fine and reports the correct temps. The problem is after a print has started, as soon as the part cooling fan connected to Fan1 kicks on it causes the ez31865 to error out with "mcu 'mcu' shutdown: thermocouple reader fault" 
I can trigger the fault at any time by starting the part cooling fan. Even worse if Fan1 is running at 100% there is no problem. It only triggers the fault if Fan1 power is between 0 and 100%.

Freddyb888, 2023-04-24:
Did you fix it? Cause I want to install same Hardware.

Michael-Jacobsen, 2023-04-24:
> Did you fix it? Cause I want to install same Hardware.

No, I am still talking with BTT about it. Maybe I just have a defective board or 31865 but It seems like a really odd fault to me.

22chrs, 2023-04-25:
If they would only share the schematics we could be able to fix these problems somehow. I am planing to buy a couple of these boards because they are perfect for my industrial project. But without the schematics it is quiet bad ..

Freddyb888, 2023-04-27:
If they would only open this site 😉

Metzlmane, 2023-04-29:
@Michael-Jacobsen Would you be so kind and share your config? I have the same error but no temperature is shown

Michael-Jacobsen, 2023-04-29:
> @Michael-Jacobsen Would you be so kind and share your config? I have the same error but no temperature is shown

What part would you like? The whole thing or just the sensor? This is the sensor in  my extruder config. I am currently using the TH0 header so it is uncommented and the max31865 is commented.
##	PT1000
sensor_type: PT1000
sensor_pin: PB0       #TH0 position
pullup_resistor: 2200
## MAX31865
#sensor_type: MAX31865
#sensor_pin: PD6     #Motor8
#spi_software_sclk_pin: PE12
#spi_software_mosi_pin: PE14
#spi_software_miso_pin: PE13
#rtd_nominal_r: 1000
#rtd_num_of_wires: 2
#rtd_reference_r: 4300

Michael-Jacobsen, 2023-05-15:
Made the mistake of buying a new EZ31865 and trying it on all other motor ports.

Of course its exactly the same. This board is going in the bin. I'll never buy BTT again.
I should have stuck with my Spider

The only "support" you get is idiotic suggestions to try things you have already told them you tried, followed by being completely ignored.
Beware this company.

gianguardo, 2023-06-27:
Hi, I have the same problem but inserting your configuration it tells me: MCU 'mcu' shutdown: Thermocouple reader fault
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

sensor_type: MAX31865
sensor_pin: PD6 #Motor8   PT100
spi_software_sclk_pin: PE12
spi_software_mosi_pin: PE14
spi_software_miso_pin: PE13
rtd_nominal_r: 100
rtd_num_of_wires: 2
rtd_reference_r: 430

Do you have any suggestions. Thank you

Michael-Jacobsen, 2023-06-28:
Hi Gianguardo

Yes that is the same error I get. I can get the sensor to work until the part cooling fan is turned on. It seems to be an SPI signalling issue. I also can't get SPI motor drivers to work while Uart drivers work fine. I would suggest connecting your sensor directly to the board just to test it. It may be that it just won't work over the distance to the toolhead.

gianguardo, 2023-06-28:
hello, I almost solved it by switching to marlin, now I managed to get EZ31865 to work but as I insert the SDCard it tells me err and the temperatures detected after 200°C stabilize in a range of 10°C. So it should be a configuration issue, I'm trying to fix it

Kashyyy24, 2023-07-11:
I finally got mine working. The onboard SD on the SPI BUS was causing issues. Add the below lines to printer.cfg to set the onboard SD CS pin to low to prevent it affecting the SPI.

`[static_digital_output my_output_pins]
pins:!PB12`

FYI - My setup has 2xEZ5160, 3xEZ2209 and 1xEZ31865. 
EZ31865 DIP switches set to '1-ON,2-ON,3-ON,4-OFF' for a 2 wire PT100.
For 3 wire you need to remove the 0 ohm link on the EZ31865 and solder the 2 contacts and shown on the right image
[image]

gianguardo, 2023-07-11:
Hi, I tried to set low Sd detect but it doesn't work, I can print via lcd (after initializing EPPROM as I receive temp err) and via usb but if I use the SDcard via the card it doesn't detect it.
Should welding also be done with a 2-wire PT100? if so which photo of the 2?
#if SD_CONNECTION_IS(ONBOARD)
  #ifndef SD_DETECT_STATE
    #define SD_DETECT_STATE HIGH  LOW
  //#elif SD_DETECT_STATE == LOW
  // #error "BOARD_BTT_OCTOPUS_MAX_EZ onboard SD requires SD_DETECT_STATE set to HIGH."
  #endif
  #define SDSS PB12
  #define SD_SS_PIN SDSS
  #define SD_SCK_PIN PE12
  #define SD_MISO_PIN PE13
  #define SD_MOSI_PIN PE14
  #define SD_DETECT_PIN PB13
  #define SOFTWARE_SPI
Thank you and good day

gianguardo, 2023-07-11:
Hi, I use the EZ5160 in spi mode, I also tried disabling TMC_USE_SW_SPI in Configuration._adv.h but it still doesn't detect my sdcard.

Kashyyy24, 2023-07-11:
@gianguardo can you share your Marlin config files? Also what LCD are you using? can you share a picture of your board setup?

gianguardo, 2023-07-11:
Enabling #define SDCARD_CONNECTION ONBOARD in Configuration_adv.h (recommended on github marlin) works fine. Thank you

Kashyyy24, 2023-07-11:
Great news. The issue comes from undefined SPI devices on the SPI BUS. All devices must be either configured for use, removed or have their CS pin set low

gianguardo, 2023-07-11:
On marlin you have to set it to :
#define SD_DETECT_STATE HIGH
that's how it works

Michael-Jacobsen, 2023-07-12:
> I finally got mine working. The onboard SD on the SPI BUS was causing issues. Add the below lines to printer.cfg to set the onboard SD CS pin to low to prevent it affecting the SPI or alternatively just ensure there is no SD card inserted.
> 
> [static_digital_output my_output_pins] pins:!PB12
> 
> FYI - My setup has 2xEZ5160, 3xEZ2209 and 1xEZ31865. EZ31865 DIP switches set to '1-ON,2-ON,3-ON,4-OFF' for a 2 wire PT100. For 3 wire you need to remove the 0 ohm link on the EZ31865 and solder the 2 contacts and shown on the right image [image]

Thankyou for sharing this. I haven't tried it with my ez31865 but it got my ez5160 drivers to work after many days of hair pulling. I had seen the klipper docs about setting static digital outputs on spi devices but I had the exclamation point missing for the sd card slot pin.
I suspect my ez31865 will work now too. I'll try it later and report back.

Michael-Jacobsen, 2023-07-14:
I still can't get EZ31865's to work. I just get constant thermocouple faults or reading of -250C very briefly. I have double checked that I have the Dip switches correct and its a PT1000 (though tried PT100 too) sensor. I tried combinations of static pins set high and low but it doesn't change anything.

Kashyyy24, 2023-07-14:
@Michael-Jacobsen Can you share your printer.cfg, klippy log and pic of your setup? how many other SPI devices do you have? have you tried other motor slots?

Michael-Jacobsen, 2023-07-14:
[klippy (2).log](https://github.com/bigtreetech/Octopus-Max-EZ/files/12047048/klippy.2.log)

The klippy log is kindof large because of all the restarting. The configs used are in the log. I have tried Motor8, Motor9 and Motor10 ports
At the moment I just have a test sensor configured and a PT1000 directly connected to the Motor ports. The other SPI devices are the 2 5160 drivers and the SD card slot
Tried with two different EZ31865 modules.
I used to be able to get this to work until the part cooling fan kicked on but now it just errors out straight away. It seems like its gotten worse.

Kashyyy24, 2023-07-14:
I didn't notice any issues in the config.
I just tested my setup in ports 8, 9 & 10 and it does seem like there is some hardware issues somewhere. For me port 8 works fine, port 9 is intermittent and port 10 doesn't work at all. I currently use port 7 and its been stable so maybe try that. 

Have you tried a simple config where only the EZ5160's and EZ31865 are connected and configured?
Just to confirm...

- Is the green power LED on the EZ31865 on? 

- Is your PT1000 connected to the 2 middle pins of the motor connector?

- Do you have another way to test your EZ31865 modules? I bought the BTT EZ-Driver Connector to easily hook it up to an Arduino to confirm it was working

- Do you have an LCD display connected?

Michael-Jacobsen, 2023-07-14:
> I didn't notice any issues in the config. I just tested my setup in ports 8, 9 & 10 and it does seem like there is some hardware issues somewhere. For me port 8 works fine, port 9 is intermittent [...]

Thankyou for looking at the config anyway. 

I'm in the middle of printing some parts for a customer so I will do some more testing soon. 
yep using the middle two pins on the 4pin motor ports. No I don't have an adaptor but that's a good idea, I might order some but that will take a couple weeks. There is no LCD connected to the octopus, I use klipperscreen through a pi4
I honestly haven't noticed if an led light comes on because the electronics are in the basement of the Voron 2.4 so its always tipped back upright before I reconnect power. I will figure a way to give it power while I can see.

Thankyou for the tip about the ports. I will try shuffling some steppers around and see if anything improves when I get a chance.

tavalin, 2023-08-24:
@Michael-Jacobsen did you manage to get this working? I'm suffering the same issue with my 2-wire PT1000 and setting 
```
[static_digital_output my_output_pins]
pins:!PB12
``` 
doesn't seem to help

Michael-Jacobsen, 2023-08-25:
> @Michael-Jacobsen did you manage to get this working? I'm suffering the same issue with my 2-wire PT1000 and setting
> 
> ```
> [static_digital_output my_output_pins]
> pins:!PB12
> ```
> 
> doesn't seem to help

Unfortunately not. I got my ez5160 drivers to work but the ez31865 don't work. At first I could get it to work until the part cooling fan kicked on but now the system errors out immediately. I have tried two different 31865 modules and in all the different ports. I think some boards just aren't as good as others. I have bought a CAN toolhead to get around it but haven't installed it yet.

meyitangp, 2023-09-16:
Has anyone figured out the correct printer.cfg settings when using the Octopus EZ V1 with EZMAX31865 chip and 2 pin PT100 sensor element connected to Motor 8 through Motor 10?? I've tried all suggestions on this thread and still get error "**MCU 'mcu' shutdown: Thermocouple reader fault**". The only thing connected to the board is the EZMAX and usb power (no mains power). I find it odd that this company sells and aggressively advertises these products but does not outline basic configuration settings.

Support is useless and non-responsive.
Attaching pictures of the board shortly.
Currently testing using Motor8.
[image]

`[static_digital_output my_output_pins]
pins:!PB12

# Motor-5
[extruder]
step_pin: PB5
dir_pin: PB4
enable_pin: !PB6
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PF6 # HE0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 200
###############################
#EZMAX31865 CONFIGURATION
###############################
sensor_type: MAX31865 #   One of "MAX6675", "MAX31855", "MAX31856", or "MAX31865".
sensor_pin: PD6 #Motor 8
spi_speed: 4000000
spi_software_sclk_pin: PE12
spi_software_mosi_pin: PE14
spi_software_miso_pin: PE13
#tc_type: K
#tc_use_50Hz_filter: False
#tc_averaging_count: 1
rtd_nominal_r: 100
rtd_reference_r: 430
rtd_num_of_wires: 2
rtd_use_50Hz_filter: True`

pav67, 2023-12-26:
Hello @meyitangp, 
Did you get it to work enventually ? Did you get any answer from support ?

meyitangp, 2023-12-26:
@pav67 yes, I had to use hardware SPI. for each motor, simply comment out the 3 software SPI lines and add “spi_bus: spi4” in the same section without quotes.

pav67, 2023-12-27:
@meyitangp thanks for your answer, I managed to have it work too. Here is my config for motor-5 : 

[extruder]
step_pin: PB8
dir_pin: !PB9
enable_pin: !PB7
microsteps: 16
rotation_distance: 3.433
#gear_ratio: 7:1
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PF6
max_extrude_cross_section: 1.3
control: pid
pid_Kp: 13.657
pid_Ki: 0.349
pid_Kd: 133.670
pressure_advance: 0.02
pressure_advance_smooth_time: 0.020
# temp sensor config :
sensor_type: MAX31865
spi_bus: spi4
rtd_nominal_r: 100
rtd_reference_r: 430
rtd_num_of_wires: 2
rtd_use_50Hz_filter: True
sensor_pin: PG10
min_temp: -20
max_temp: 350