bigtreetech/Octopus-Max-EZ #9: VOT USB not sending enough power to RPi4

Bingkokek, 2023-05-16:
I am powering my RPi4 through the VOT USB A port on the Octopus Max EZ but my RPi keep on reporting undervoltage. I thought the USB port is supposed to have enough power to power a Raspberry Pi. Any ideas on what is happening?

lianghongkey, 2023-05-26:
Does usb cable is too fine?

IceDragonix, 2023-05-28:
Same problem here. Just built a new printer with OCTOPUS MAX EZ and Raspberry Pi 4B. I receive insufficient voltage messages and throttle warnings from RPi.

IceDragonix, 2023-06-08:
> Same problem here. Just built a new printer with OCTOPUS MAX EZ and Raspberry Pi 4B. I receive insufficient voltage messages and throttle warnings from RPi.

In my case it was indeed the cable, or rather the USB-A to USB-C adapter I was using. I replace the cable with standard USB-A to USB-C and the problem is no longer present.

jonathanperret, 2023-06-26:
I can confirm I have no issue either powering a RPi4 through the Octopus MAX EZ's USB-A port. I'm using a very short (10cm) A-to-C cable, which might help.

Btw, I saw someone (probably on YouTube, can't find it right now) who seemed to be using one A-to-C cable to power the Pi from the Octopus, and another A-to-C cable between the Pi and the Octopus for the USB communication. This does appear to work, but you can actually get rid of that second cable by enabling USB data traffic on the Pi's USB-C port: https://techoverflow.net/2022/04/19/how-to-enable-usb-c-host-mode-on-raspberry-pi/

I have it all working now with that single USB A-to-C cable: the Octopus is powering the Pi through that cable (the VUSB jumper is in place) but the Pi is the USB host and enumerates the board correctly. This feels quite weird but so far nothing has gone up in flames (after all, the Pi's USB-C port is _designed_ to power the Pi…). If this configuration is actually supported, it would be nice to have it described in the manual.

Metzlmane, 2023-06-27:
what is the maximum current for the 5V before the fuse triggers? I can't find this anywhere in the so called "documentation"

jonathanperret, 2023-06-27:
According to the published schematic, the 5V fuse is a STEF05D configured with a 15R resistor. I'm not experienced enough with these devices to be sure I understand the datasheet correctly, but this seems to be setting the current limit around 4 amperes. This should be ample to supply a Raspberry Pi 4 (which likes 3A), however the 5V line is used all over the place on the Octopus so YMMV ?

[image]

Michael-Jacobsen, 2023-07-14:
It does seem like there is some issues of some type with power. I get intermittent warnings of low power to the Pi4. I bought a bunch of different short 15cm usb cables and I can swap one in, it works fine for a while then it starts the warnings again.

sargonphin, 2024-01-03:
> I can confirm I have no issue either powering a RPi4 through the Octopus MAX EZ's USB-A port. I'm using a very short (10cm) A-to-C cable, which might help.
>
> Btw, I saw someone (probably on YouTube, [...]

Okay man, I need to thank you a lot about your post here. I was running into the same USB-A issues regarding power delivery, and it turns out that high-power (amperage) is only available on the Type-C of the board. I enabled the host-mode on the type-C and solved all power related problems on top of saving one cable off of the printer box :D top 100% answer, thank you so much!

sargonphin, 2024-01-03:
We have done some tests now.

I would like to add that the 5V 5A that is being advertised is not about the USB power, but the overall 5V rail that the board offers (for LED's, among other things). There is NOT 5A available on the USB ports, not at all! Our tests showed that a maximum of 1.5A is available (but the low-voltage alarm will trigger at around 1.2A, although voltage is still within acceptable USB voltages, the Raspberry is just being conservative)

- 1.5A is the standard for USB-C single-lane that don't support any form of Power-Delivery (PD) so as far as BigTreeTech is concerned, everything is correct

- The USB type-A port is delivering a maximum current of 500 mA before voltage drop, which again is on-par with the USB 2.0 Type-A standard

- A workaround is to use one of the 5V addressable LED output ports as these can handle as maximum of 3A per plug and connecting the power lines to the GPIO of the Raspberry

So there is no problem with the board. The link to the trick @jonathanperret posted will fix all problems related to USB power 👍