bigtreetech/Octopus-Max-EZ #12: Problem with upload firmware via SD card

thral666, 2023-06-02:
Hi,

I have a problem with upload firmware by SD card. I do everything as in the user manual but when I power the board, only one red LED (near reset button) is on continuously and nothing happens :(

SD card: SanDisc ultra SD HC, 8Gb, FAT32, 

[image]

ZainHan33, 2023-06-09:
I had a similar problem and trying a different sd card worked for me

Michael-Jacobsen, 2023-06-28:
I use a 32Gb SD card on my board and it works fine. Sometimes you just have to try different cards, the capacity isn't the problem. Its becoming increasingly difficult to buy cards smaller than 32Gb because they are so cheap.

math45klm, 2023-07-11:
Hi! just a small precision.. for my Octopus Max EZ, the 8gb SD card flashed without problem 👌

ebraiman, 2023-09-23:
If you're having issues with Flashing with firmware.bin, create a second file named FIRMWARE.CUR and see if it works.