bigtreetech/Octopus-Max-EZ #13: Power on board via USB

22chrs, 2023-06-04:
Powered my board only via the USB-C connection of the board. The power LED is not flashing red and there is not device in the serial list. Jumper cap onto the VUSB. What's wrong?

jonathanperret, 2023-06-19:
Try another USB-C cable (the one that was provided in the box worked for me). I initially tried to power the board using an Apple USB-C to USB-C cable and no voltage was coming through, as you describe. It looks like the board's USB-C sink is not fully spec-compliant and a "too smart" cable + PSU combination will refuse to power it.

jonathanperret, 2023-06-26:
The schematic confirms this board has the same issue as the Raspberry Pi 4 : https://github.com/bigtreetech/Octopus-Max-EZ/issues/4#issuecomment-1607243772