bigtreetech/Octopus-Max-EZ #14: does not see /dev/serial/by-id/????

KumphanartDansiri, 2023-06-05:
In old version of RPI3 and windows, I can see /dev/serial/by-id as per klipper instruction.  In RPI4, cannot see that directory.
Once flash klipper.bin to Octopus Max EZ, the flash failed.  Now no more /dev/serial/by-id showing up.  Only if press DFU Mode, the board does show up as DFU in FS Mode, but no more response in normal boot.

Any default bootloader / firmware to flash with STCubeProgrammer is highly appreciate.

22chrs, 2023-06-05:
@KumphanartDansiri Have you worked it out to upload a custom firmware to the board?
I modified the marlin code and I am successfully able to compile my code. Sadly I can't upload it via serial. 
I am not getting the device to show up as serial device.

KumphanartDansiri, 2023-06-07:
Just Press the middle button from the 3 button array in the middle of the board during boot.  This will put the board into DFU mode.  Then, you can use STM32CubeProgrammer to connect in DFU Mode and upload the firmware to the board.  I did successfully upload klipper to the board but still does not see serial port.

KumphanartDansiri, 2023-06-07:
@bigtreetech Can you provide original bootloader or firmware that come with the board so I regain function of the serial port.

KumphanartDansiri, 2023-06-17:
Solved.

User command "make flash FLASH_DEVICE=0483:df11"  should flash at address 0x0000000. 
Or use Cube Programmer to load the file and everything work.

closed 2023-06-17 by KumphanartDansiri