bigtreetech/Octopus-Max-EZ #19: Octopus EZ V1 and EZ31865 klipper configuration not working.

meyitangp, 2023-10-04:
Has anyone figured out the correct printer.cfg settings when using the Octopus EZ V1 with EZMAX31865 chip and 2 pin PT100 sensor element connected to Motor 8 through Motor 10?? I've tried all suggestions on this thread and still get error "MCU 'mcu' shutdown: Thermocouple reader fault". The only thing connected to the board is the EZMAX and usb power (no mains power).
[image]

Support is non-responsive.
Currently testing using Motor8.
EZ issues

`[static_digital_output my_output_pins]
pins:!PB12

Motor-5
[extruder]
step_pin: PB5
dir_pin: PB4
enable_pin: !PB6
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PF6 # HE0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 200
###############################
#EZMAX31865 CONFIGURATION
###############################
sensor_type: MAX31865 # One of "MAX6675", "MAX31855", "MAX31856", or "MAX31865".
sensor_pin: PD6 #Motor 8
spi_speed: 4000000
spi_software_sclk_pin: PE12
spi_software_mosi_pin: PE14
spi_software_miso_pin: PE13
#tc_type: K
#tc_use_50Hz_filter: False
#tc_averaging_count: 1
rtd_nominal_r: 100
rtd_reference_r: 430
rtd_num_of_wires: 2
rtd_use_50Hz_filter: True`

math45klm, 2023-10-04:
hi, have you tried with the jumper to select the 24v associated with the port concerned?

meyitangp, 2023-10-06:
@math45klm Yes I have. It made no difference. What is the pinout for Motor8 socket?

math45klm, 2023-10-07:
I also have the ez31865 to install for a pt1000 2 wire but I have not yet installed and I have only just read that the ez31865 uses VCCIO part of the power supply, support 3.3v-5v power input.. I deduce from this that you should not install the jumper for the 24v (perhaps wrongly) I will try to see things more clearly before making you make more mistakes, I am not sure sorry. I will come back to you if I managed to install it on my side.

meyitangp, 2023-10-07:
@math45klm Ok thank you. I will try to trace MCU pins to determine Motor 8 Socket pinout. It would be nice if BIQU could provide this information. Motor socket pinout from MCU.

meyitangp, 2023-10-15:
Still not able to get this work. Waiting on support.

meyitangp, 2023-10-20:
for anyone who faces this issue in the future: simply comment out the 3 software spi lines and use “spi_bus: spi4”.

closed 2023-10-20 by meyitangp

egoleos, 2023-11-07:
> for anyone who faces this issue in the future: simply comment out the 3 software spi lines and use “spi_bus: spi4”.

it doesn't work with tmc5160 steppers. can't use spi_bus: spi4 in the settings.