bigtreetech/Octopus-Max-EZ #23: No trigger on x after full movement

DammyBod, 2024-03-21:
I am having problem with homing all my stepper motors. Although I have three motors connected to [stepper_x], [stepper_y], and [stepper_z] respectively. The initial error I got was "Endstop x still triggered after retract." After checking for possible solutions online, I made changes to my endstop_pin variables. For [stepper_x], I changed from PF0 to ^!PF0, I added the same characters to the variables for [stepper_y] and [stepper_z]. I also changed the position_endstop to the maximum position for both X and Y axes then I started getting "No trigger on x after full movement."

Kindly suggest the best solution because I have my pinout variables correctly set just as it was shown on the pinout diagram for the  Octopus Max EZ.

[image]
[image]

DammyBod, 2024-03-21:
This is what my connection looks like at the moment.
[image]
[image]

cadamwil, 2025-02-18:
You don't have any endstop switches hooked up, nor do you have Sensorless homing defined.  You need to choose one or the other and configure appropriately.